QUaServer/QUaServer #4: Please document how to add (child) nodes with string ids

eudoxos, 2019-07-12:
This is a MWE where I try to create object hierarchy with user-defined string ids (rather than random numerical ids created by the server). Adding a folder and a variable works fine, but adding a nested variable inside the folder fails (crash). If the folder does not explicitly define its own id, adding the nested variable works fine.

```
#include <QCoreApplication>
#include<QUaServer>
int main(int argc, char** argv){
   QCoreApplication app(argc, argv);
   QUaServer server;
   QUaFolderObject *root=server.objectsFolder();
   QUaFolderObject* folder=root->addFolderObject("ns=1;s=myfolder");
   QUaBaseDataVariable* var=root->addBaseDataVariable("ns=1;s=myvar");
   QUaBaseDataVariable* nestedVar=folder->addChild<QUaBaseDataVariable>("ns=1;s=mynestedvar"); // same as folder->addBaseDataVariable(...)
   server.start();
   return app.exec(); 
}
```

Tracing the code in quaserver.h https://github.com/juangburgos/QUaServer/blob/d9851da032e4c04a8a996ac02b41deb4ec040136/src/wrapper/quaserver.h#L448-L463, I see that createInstance returns a non-null id (`UA_NodeId_isNull` returns false) but `QUaNode::getNodeContext` returns `nullptr` which makes `Q_CHECK_PTR` fail.

Maybe I am short of understanding how nodes are supposed to be named in OPC/UA. Are folders not supposed to have string ids? Do nested variables need to have the parent node (folder) id as a part of their own id?

juangburgos, 2019-07-12:
Hi,
It might be related to a [bug I found in open62541](https://github.com/open62541/open62541/issues/2667). What I found was that I couldnt create an instance of a custom type with a custom NodeId. Quite similar that what you describe but you don't use custom types.
I will check your example to try find out if it is the same issue.

juangburgos, 2019-07-12:
Ite seems it was related to that bug, but it was not an open62541 bug but a mistake of my own. 

I did not know that cleaning the browsing results of the server's browse service also cleaned the resulting NodeIds.

Therefore when browsing for parent, or children or any other reference that has a non-numeric NodeId, then the result was always null or invalid.

I believe now I fixed it in all the required places. Please reopen if issue persists.

commit 69691211 (2019-07-12): Fixes #4
closed 2019-07-12 by juangburgos