QUaServer/QUaServer #5: Certificate and Encryption

embbo, 2019-07-18:
Hi all, I am new to encryption and certificate .. I builed accroding to tutorials given on this page. but I have some question . 

1. when I used "ca.der.crl" and "ca.crt.der" and try to connect to with Server it connects fine but if  i delted these two files and try tou connect UaExpert with Server .It shows me some warning and I have to Accerpt the server .. and again ot connects ..So my question is what is difference between both cases.
[image]

juangburgos, 2019-07-18:
Hi,
Did you already read [this document](https://github.com/juangburgos/QUaServer/blob/master/VALIDATION.md)? Do you understand the concepts in there?

You are getting those red warnings because the client needs to know in advance which servers he can trust. That information is included in the Certificate Authority's certificate and revocation list.

embbo, 2019-07-18:
yes i did, but my question is why it connects when I check the option ..... Accept the server certificate for this session . (at the end of scree shot).

juangburgos, 2019-07-18:
It connects because you are explicitly saying that you trust the server. 

So the client app allows you to continue eventhough it does not have an automatic way to verify the trustworthyness of the server, but that is only because you manually chose to trust it.

Have you ever seen these in your browser?

[image]

Is the same thing, the browser is telling you "I have no automatic way to determine that this website is trust worthy, but you can continue if you want"

embbo, 2019-07-18:
okay , got it , it means it allowed from server side , that it can connect to any client

juangburgos, 2019-07-18:
No, is the client that has to trust the server. The server allows for any client to connect by default, unless you enable [user/password authentication](https://github.com/juangburgos/QUaServer#users). 

Do not confuse the mechanism that make clients trust servers with the mechanims that make servers trust clients.

embbo, 2019-07-18:
okay ... let me read it again come to you back

embbo, 2019-07-22:
okay, It works fine thank you ...but still i have a question . Can I write these key.der and cer.der in eeprom ? at run time ..

juangburgos, 2019-07-22:
If you'd like to have those files embedded in your application I suggest to bundle them using [The Qt Resource System](https://doc.qt.io/qt-5/resources.html). This way your app will bundle the files with the executable, but in code you can treat them as regular files.

embbo, 2019-07-22:
but I mean at he run time? so i can direct upload from node ..

juangburgos, 2019-07-22:
I mean the server files (`server.crt.der` and `server.key.der`), they need to be loaded once at startup and that's it. There is no need to load them afterwards at run time. So you can use The Qt Resource System for those. 

The client files though (`ca.crt.der`, `ca.der.crl`), those need to be installed manually for every client, there is no automated way to do that.

embbo, 2019-07-22:
okay , I got it thank you :)

closed 2019-07-22 by juangburgos

embbo, 2019-07-22:
but Is it possible , If i want to change the certificate at run time and want to write in EEPROM .

juangburgos, 2019-07-22:
That I dont know. I dont handle EEPROM. You will have to investigate yourself. Good luck and let me know if you succeed.

embbo, 2019-07-22:
okay thnakyou , I will :)

embbo, 2019-07-24:
can I used " -----BEGIN CERTIFICATE-----" and -----BEGIN PRIVATE KEY----- as string instead of Hex  0x2D 0x2D 0x2D  from server.crt and server.key file rspectively