QUaServer/QUaServer #6: new amalgamation does not compile

eudoxos, 2019-07-18:
Since the amalgamation was updated in https://github.com/juangburgos/QUaServer/commit/4249317bb8291ee0ad47f53dbae4011e0a0ad32d, I am not able to compile -- `UA_SOCKET` is undefined. Am I missing something? The older version compiles just fine. The platform is Ubuntu 18.04 with gcc 7.4. The errors are below. For now, I am sticking to the older version, though someone else might see the same issue as well...

```
../src/other/QUaServer/src/amalgamation/open62541.h:23426:5: error: ‘UA_SOCKET’ does not name a type; did you mean ‘UA_FORMAT’?
     UA_SOCKET sockfd;              /* Most connectivity solutions run on
     ^~~~~~~~~
     UA_FORMAT
../src/other/QUaServer/src/amalgamation/open62541.h:28242:17: error: ‘UA_SOCKET’ was not declared in this scope
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
                 ^~~~~~~~~
../src/other/QUaServer/src/amalgamation/open62541.h:28242:17: note: suggested alternative: ‘UA_FORMAT’
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
                 ^~~~~~~~~
                 UA_FORMAT
../src/other/QUaServer/src/amalgamation/open62541.h:28242:35: error: expected primary-expression before ‘const’
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
                                   ^~~~~
../src/other/QUaServer/src/amalgamation/open62541.h:28242:59: error: expected primary-expression before ‘len’
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
                                                           ^~~
../src/other/QUaServer/src/amalgamation/open62541.h:28242:64: error: expected primary-expression before ‘int’
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
                                                                ^~~
../src/other/QUaServer/src/amalgamation/open62541.h:28242:73: error: expression list treated as compound expression in initializer [-fpermissive]
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
                                                                         ^
../src/other/QUaServer/src/amalgamation/open62541.h:28246:16: error: ‘UA_SOCKET’ was not declared in this scope
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
                ^~~~~~~~~
../src/other/QUaServer/src/amalgamation/open62541.h:28246:16: note: suggested alternative: ‘UA_FORMAT’
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
                ^~~~~~~~~
                UA_FORMAT
../src/other/QUaServer/src/amalgamation/open62541.h:28246:34: error: expected primary-expression before ‘const’
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
                                  ^~~~~
../src/other/QUaServer/src/amalgamation/open62541.h:28246:58: error: expected primary-expression before ‘len’
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
                                                          ^~~
../src/other/QUaServer/src/amalgamation/open62541.h:28246:63: error: expected primary-expression before ‘int’
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
                                                               ^~~
../src/other/QUaServer/src/amalgamation/open62541.h:28246:74: error: expected primary-expression before ‘const’
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
                                                                          ^~~~~
[...]
```

juangburgos, 2019-07-18:
I generate the included amalgamation on windows using the **cmake** command:

```
cd open62541.git
mkdir build; cd build;
cmake -DUA_ENABLE_AMALGAMATION=ON .. -G "Visual Studio 15 2017 Win64"
```

In the past, then open the VS project and build it, which also generates the amalgamation files. I copy the amalgamation files to my QUaServer repo and then I use [this **QMake** project](https://github.com/juangburgos/QUaServer/blob/master/src/amalgamation/open62541.pro) to build the amalgamation separately as a library.

With previous amalgamations it turned out that they also compiled on linux out of the box. It might be that newer amalgamations generated on windows do not compile anymor eon Linux. So you might have to generate your own amalgamation on Linux and replace the one provided in this repo.

You could try that out and let me see if it works, then we might also check what differences are between Windows and Linux amalgamations.

Anyways I might be updating the amalgamation again soon, once [this issue's fix](https://github.com/open62541/open62541/issues/2866) is merged into the open62541 master branch.

I am sorry I cannot guarantee the amalgamation will work in all platforms, but I am low in resources. Maybe when I have the time, I will just add open62541 as a git submodule and add a QMake command to call CMake and generate the amalgamation automatically for each platform. Until then I guess we will keep having issues like this one.

eudoxos, 2019-07-18:
The difference is that the header in the new amalgamation is missing the posix bit of the older version, with this leader comment:

```
/*********************************** amalgamated original file "C:/Users/User/Desktop/Repos/open62541.git/arch/posix/ua_architecture.h" ***********************************/
```
Strangely enough, CMakeLists.txt for open62541 did not change recently https://github.com/open62541/open62541/blame/master/CMakeLists.txt (the `UA_AMALGAMATION_ARCHITECTURES` option has the default `win32;posix`).

juangburgos, 2019-08-01:
Worked towards solving this issue in [this commit](https://github.com/juangburgos/QUaServer/commit/a8d6866aff165fd5890e9e70e9e120799986232f). 

I removed the included amalgamation and, on Windows, now the amalgamation is built automagically when running qmake on the [amalgamation project](https://github.com/juangburgos/QUaServer/blob/a8d6866aff165fd5890e9e70e9e120799986232f/src/amalgamation/open62541.pro#L19).

All the magic goes in [open62541amalgamation.pri](https://github.com/juangburgos/QUaServer/blob/a8d6866aff165fd5890e9e70e9e120799986232f/src/amalgamation/open62541amalgamation.pri#L1).

Are you feeling like giving it a go to the [missing Linux implementation](https://github.com/juangburgos/QUaServer/blob/a8d6866aff165fd5890e9e70e9e120799986232f/src/amalgamation/open62541amalgamation.pri#L145)?

juangburgos, 2019-08-09:
Fixed for Linux in [this commit](https://github.com/juangburgos/QUaServer/commit/35d1d20bd4815e72b05b9f0a290cf394d317f35f). Now just missing Mac, but I dont have one.

eudoxos, 2019-08-09:
Sorry for not being helpful, had little time plus I don't use qmake and don't know much about it, I use straight CMake. I am happy Linux is supported now for others who use qmake. Thanks!

juangburgos, 2019-08-09:
No prob. I will close the issue for now, leave Mac unsupported. This ain't javascript.

closed 2019-08-09 by juangburgos