QUaServer/QUaServer #15: Provide a way to access m_server for tuning

eudoxos, 2019-11-04:
I changed configuration of the internal `UA_Server*` so that its logging goes through [spdlog](https://github.com/gabime/spdlog). I had to make `m_server` public for that purpose. Is there a way to do it with the current API?

Also, there is a message from QUaServer ctor before the logging is set up (`warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.`). Would it be sensible to have logger as additional ctor parameter so that logger is change on time or would it be too much params already?

FWIW, the current code follows, perhaps useful for someone:

```
spdlog::level::level_enum UA_LogLevel_2_spdlog_level(UA_LogLevel l){
	switch(l){
		case UA_LOGLEVEL_TRACE: return spdlog::level::level_enum::trace;
		case UA_LOGLEVEL_DEBUG: return spdlog::level::level_enum::debug;
		case UA_LOGLEVEL_INFO: return spdlog::level::level_enum::info;
		case UA_LOGLEVEL_WARNING: return spdlog::level::level_enum::warn;
		case UA_LOGLEVEL_ERROR: return spdlog::level::level_enum::err;
		case UA_LOGLEVEL_FATAL: return spdlog::level::level_enum::critical;
		default: throw std::runtime_error("Unknown UA_LogLevel "+std::to_string(l));
	}
};

struct UA_log_spdlog_context{
	std::shared_ptr<spdlog::logger> logger;
};

void ua_log_spdlog_callback(void* context, UA_LogLevel level, UA_LogCategory category, const char *msg, va_list args){
	UA_log_spdlog_context* ctx=(UA_log_spdlog_context*)context;
	auto spdlev=UA_LogLevel_2_spdlog_level(level);
	if(!ctx->logger->should_log(spdlev)) return;
	const char *logCategoryNames[7] = {"network","channel","session","server","client","userland","securitypolicy"};
	char buf[255];
	std::vsnprintf(buf,sizeof(buf),msg,args);
	ctx->logger->log(spdlev,"({}) {}",logCategoryNames[category],buf);
};

void ua_log_spdlog_clear(void* context){
	((UA_log_spdlog_context*)context)->logger.reset();
};

shared_ptr<spdlog::logger> open62531_logger=spdlog::stdout_color_mt("open62531");
UA_log_spdlog_context ua_log_spdlog_context;
UA_Logger spdlog_ua_logger;

/* starting routine */
server=new QUaServer(/*port*/port,/*certificate*/QByteArray(),/*parent*/this);
ua_log_spdlog_context.logger=open62531_logger;
spdlog_ua_logger.log=&ua_log_spdlog_callback;
spdlog_ua_logger.context=&ua_log_spdlog_context;
spdlog_ua_logger.clear=&ua_log_spdlog_clear;
UA_Server_getConfig(server->m_server)->logger=spdlog_ua_logger;
```

juangburgos, 2019-11-05:
Yes. It is something I thought of adding to the API. Nice to see you got it working. I still have not decided on how to add it. A callback is a possibility, but as you mentioned it might clutter the constructor. Maybe define it as static? But then in the unlikely case many servers are instantiated, how to differentiate which logs belong to which server?

I have been meaning to add many things to the API, but currently I have been too busy at work. I hope to be able to contribute more soon.

juangburgos, 2019-12-23:
I was just revisiting this and I see two options to avoid adding parameters to the contructor:

* Enable logging redirection be default to a `QUaServer` signal. So the user would have to connect to the signal explicitly and print the message to wherever they like.

Pros: No params to constructor. A Qt-way of doing things (signals).

Cons: We might miss the first message you mentioned about `leak credentials`. Also the user would see no log be default if he did not connect to the signal.

The second Con might be mitigated by only redirecting the log to the signal if there is more than one connection to it using [`QObject::receivers`](https://doc.qt.io/qt-5/qobject.html#receivers).

* Add a virtual method to implement the custom log. The user would have to inherit from `QUaServer` and implement the method.

Pros: No params to constructor. A C++ of doing things (inheritance). The default implementation just prints to console.

Cons: We might miss the first message you mentioned about `leak credentials`. Also the user would have to use iheritance (don't know if this is a Con, but I personally do not like having to inherit just to get some simple extra functionality).

---

So, what are your thoughts?

juangburgos, 2020-01-09:
Hello there, just I heads up. I just **broke** the constructor API, decided to have argument-less constructors to avoud the temptation of clutter the arguments in the future. Also to fix some issues when stoping and restarting a server.

eudoxos, 2020-01-09:
Hi Juan, sorry I did not comment earlier.

I think argument-less constructor makes sense.

As for logging, I think providing access to the internals (the open62541 server) is okay as far as I see.

* Most logging systems are asynchronous already, thus signal/slot will not add an extra benefit in that sense, it will only be an extra layer. The user will have to, anyway, write the signal->logging system conversion for the logging system she uses, so why not write just open62541 plugin which will bypass Qt.
* I would just add something like `QUaServer::get_internal_UA_Server_pointer_DANGEROUS()` or something similar, with disclaimer to use it at one's own risk.
* the first message about leaking credentials is unavoidable as it happens in UA_Server constructor already (I read some upstream ticket about it but cannot find it ATM).

juangburgos, 2020-01-18:
Hi,

In the end I decided to have both; a Qt logging API (a Qt signal with the message wrapped in a QString) for the normal user who does not want to bother looking into the unserlying open6254lib. Then also the posibility of retrieving the internal open62541 config for power users such as you.

When looking into the code for the posibility of `get_internal_UA_Server_pointer_DANGEROUS` I realized that it might not be possible due to it being really dangerous, because every timevthe server is started, the open62541 config is wipped and created again, therefore, if I give the user a reference, it might get invalidated at any time without notice.

You can see the simple logging API I implemented. It is just a signal:

```c++
// Log API
void logMessage(const QUaLog &log);
```

The `QUaLog` is simply:

```c++
struct QUaLog
{
	QString        message;
	QUaLogLevel    level;
	QUaLogCategory category;
};
```

eudoxos, 2020-02-10:
Hi, looks good.

I am only somewhat concerned with converting each and every message to QString (whether it gets logged or not at the end). How about std::string? It manages its storage but does not do any encoding conversion and such. Or QByteArray, if you insist on Qt.

In my case, I pass it on to spdlog which uses std::string: so there is conversion to QString and then back again, with all the encoding stuff.

eudoxos, 2020-02-10:
BTW I appreciate your work on this project a lot :) Thanks.

juangburgos, 2020-02-11:
I think you are right, I did not test peformance because I left for holidays. I think I will stick to `QByteArray`, to keep it Qt-ey. I will update it soon.

I have nothing against the `std::`, Actually I would have liked to make a wrapper without Qt, sadly the standard still does not have any of the reflection features needed that Qt-moc provides. So I rather keep it all Qt-ey for consistency.

juangburgos, 2020-02-26:
Hopefully this is resolved now, if not, please re-open. I also added a [serialization API](https://github.com/juangburgos/QUaServer#serialization) that you might find useful if your application creates nodes on runtime.

closed 2020-02-26 by juangburgos

eudoxos, 2021-04-07:
As you see, I am back here. I appreciate a lot the progress you've done with QUaServer, Very nice, really.

To the point of logging, I have very simple server setup like this:
```c++
server=new QUaServer(/*parent*/this);
// note that I connect logging signals as soon as possible
QObject::connect(server,&QUaServer::logMessage,this,&OpcUAServer::logToSpdlog);
// set some server params, create a few nodes etc
server->setPort(port);
server->setApplicationName("...");
// start now
server->start();
```
and I get this in the console:
```
[2021-04-07 19:15:59.764 (UTC+0200)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-04-07 19:15:59.764 (UTC+0200)] info/server        AccessControl: Anonymous login is enabled
[2021-04-07 19:15:59.764 (UTC+0200)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2021-04-07 19:15:59.764 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
19:15:59 [ua.Server ]     <open62531>:1    [war] AccessControl: Unconfigured AccessControl. Users have all permissions.
19:15:59 [ua.Server ]     <open62531>:1    [inf] AccessControl: Anonymous login is enabled
19:15:59 [ua.Server ]     <open62531>:1    [war] Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
19:15:59 [ua.UserLan]     <open62531>:1    [war] AcceptAll Certificate Verification. Any remote certificate will be accepted.
```
the first 4 lines are logged from QUaServer directly (I guess?) while the next ones go through my logging handler (spdlog). Is this duplicity intentional? Any way to avoid the first 4 lines?

juangburgos, 2021-04-08:
Hi, this is because the QUaServer constructor initializes the open62541 server instance (with some default config) *before* the log handler can be set, so the very first messages are printed by the internal open62541 log handler.

The *duplicity* appears because internal open62541 server instance has to be *reconfigured* when the port or other server params are set, so the open62541 server sends the warnings again for the new config.

I still need to find a way handle this situation better.

eudoxos, 2021-04-08:
Alright :) #31

commit ba833391 (2020-02-26): Fixes #15