QUaServer/QUaServer #16: generating variable value on-demand

eudoxos, 2019-11-06:
I have some variables which should not be updated automatically -- the expense is considerable and they are hardly ever going to be used. Instead, I would like to define a functor computing the value on-demand, when the client asks for it. Is there a way to do it, or is there not because it is not implemented in QUaServer or in open62541 or is this beyond the OPC/UA architecture? Thanks! :)

juangburgos, 2019-11-06:
What do you mean with "*the expense is considerable*"? Is it in memory or CPU usage? If I remember correctly the *open62541* library supports two ways to handle data sources for variables:

1. Handled internally by the *open62541* library, read and write through the *open62541* API.

2. Handled externally,  read and write through user-defined callbacks.

For the *QUaServer* implementation I use the first approach.

eudoxos, 2019-11-06:
(admittedly an abuse of OPC/UA:) The program is logging messages into circular buffer which would, on-demand, generate latest messages in human-readable form and send them over OPC/UA; the access over OPC/UA would be quite rare in practice (when debugging), periodic publishing seems like waste of resources. I did not measure eaxactly how expensive it would be; likely negligible. It just feels unnecessary.

juangburgos, 2019-11-06:
I still don't fully undestand your use-case, but the phrase "*periodic publishing seems like waste of resources*" stands up. Are you subscribed throught a client to changes in the server? Because in that case you can tweak the OPC UA subscription (in the client) to control how often receive changes.

eudoxos, 2019-11-06:
Most the time the value of the node would not be read at all. Only sometimes, when debugging, the value would be accessed, at which point it should be built.

https://github.com/juangburgos/QUaServer/blob/1444848a9491e661173cadb3969167cbb26888a2/src/wrapper/quabasevariable.cpp#L50

If the callback setting the current value were added, would it get called before the built value were returned over OPC/UA in the normal (current) way?

juangburgos, 2019-11-06:
You can expriment with that, but as I mentioned, I built QUaServer based on approach 1), therefore I don't know what side effects might have implementing that callback. Other stuff might stop working. Take a look at <https://open62541.org/doc/0.3/server.html#data-source-callback>

eudoxos, 2019-11-06:
Thanks for the link. It is the [value callback](https://open62541.org/doc/0.3/server.html#value-callback) which looks what I'm after: "If not NULL, they are called **before reading** and after writing respectively." As I understand, upon reading request, the callback would get the data from the source, and set the OPC/UA value; the reading would proceed through the normal codepath afterwards. It is still approach 1., only calling the hook right before reading.

I will post my findings here.

juangburgos, 2019-11-11:
Fixed by pr #17

closed 2019-11-11 by juangburgos