QUaServer/QUaServer #18: Error BadCertificateHostNameInvalid

embbo, 2019-12-05:
[image]

Hi i am getting this error bfore i connect to server ..I am using UaExpert . Can anyone please eloborate it.

juangburgos, 2019-12-05:
* Did you follow [these instructions](https://github.com/juangburgos/QUaServer#create-certificates) completely and correctly?

* Could you please add que UaExpoert log (appears on the bottom)?

* Could you also add the server application's log (prints to the console)?

embbo, 2019-12-05:
1- Yes i follow all the steps

2- 
[image]

juangburgos, 2019-12-05:
In the `exts.txt` file did you remember to add to `subjectAltName` the name that the client uses to connect to the server?

I need more information to really see what was configured incorrectly. What happens if you click the *Ignore* button? Con you still connect?

embbo, 2019-12-05:
1. [v3_ca]
subjectAltName=DNS:localhost,DNS:ppic09,IP:127.0.0.1,IP:192.168.10.162,URI:urn:unconfigured:application
basicConstraints=CA:TRUE
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid,issuer
keyUsage=digitalSignature,keyEncipherment
extendedKeyUsage=serverAuth,clientAuth,codeSigning

2- yes if i ignore it, it connects to server

juangburgos, 2019-12-05:
The `ppic09` part is an example, but it must be your computer's name. You have to put *your* computer name. What troubles me from the log is the `opc.tcp://l:4840` part, is your computer's name `l`?

So if you *Ignore* then it connects, but does not print any other useful information?

embbo, 2019-12-05:
opc.tcp://l:4840 behind is ip adress.  'opc.tcp://192.168.10.162:4840' i am using embedded device so i put ip adress also there

embbo, 2019-12-05:
yes it connects without showing any problem

juangburgos, 2019-12-05:
You **must** also put the computer name in `exts.txt`

embbo, 2019-12-05:
okay let me try this... and will comeback to you

embbo, 2019-12-05:
okay I put my computer name in it and it works fine thank you ...but i still have question why validity of certificate is 1 month ?

juangburgos, 2019-12-05:
You can change the validity when you create it. It can be 1 month, 1 year or 100 years. Set it in the command:

```bash
# instead of : openssl req -new -x509 -days 3600 ...
openssl req -new -x509 -days 360000 ...
```

Please consider giving a github **star** to this project.

closed 2019-12-05 by juangburgos