QUaServer/QUaServer #20: Users with ecryption

superlambchops, 2019-12-23:
I've successfully run both the user and encryption examples. If I add the following lines to the encryption example:
	server.setAnonymousLoginAllowed(false);
	server.addUser("juan", "pass123");
	server.addUser("john", "qwerty");
I get "Error 'BadIdentityTokenInvalid' was returned during ActivateSession" in the UaExpert log and no connection.

juangburgos, 2019-12-23:
Hi, well that is exactly what is supposed to happen and is actually exactly documented like that in the `README.md` file

* <https://github.com/juangburgos/QUaServer#users>

Unless I am missing something, I believe the behaviour is correct.

juangburgos, 2019-12-23:
Sorry, my bad. I reproduced it and you were right. That was  bug there but is fixed now on master.

commit 1ced2481 (2019-12-23): Fixes #20 and also CRL expiry time
closed 2019-12-23 by juangburgos