QUaServer/QUaServer #21: server.key.der OR server.crt.der of same size

embbo, 2020-01-13:
Hi all, 
Is this possible to creat (server.key.der OR server.crt.der) of same size .. 1024 or 2048.
I fond that, it changes the size everytime when ever  i creat the cerificates and then  I have to change the buffr size. Is there any parameter from where i can fix the size of certificates .

juangburgos, 2020-01-14:
Sorry, the creation of certificates is out of the scope of this project. This project only supports loading existing certificates. To get an answer to your question, I would suggest an stackoverflow question regaring the usage of the `openssl` utility. Good luck.

closed 2020-01-14 by juangburgos

embbo, 2020-01-14:
https://github.com/juangburgos/QUaServer#server 
Hi again ,I am talking about this.. i

juangburgos, 2020-01-14:
`QUaServer` does not care about the size of the certificates. You just call the `void       setCertificate(const QByteArray& byteCertificate);` method. 

The link you posted just give some general guidelines for creating certificates. But any advanced certificate creation is out of the scope of this project. You have to find out yourself. Good luck.