QUaServer/QUaServer #22: Add API to set node timestamp (last update)

eudoxos, 2020-02-28:
I am setting a number of node values, related to one single event. It would be useful (I am actually requested to do that) to set one exact timestamp for all values updated for that event. Is there a way to do it with the current API?

It is *not* necessary to update all nodes simultaneously (like atomic transaction): while values are updated, some nodes may have old and some already new value, but the timestamp for all new (updated) nodes should be the same.

I was only looking for setTimestamp or similar, by text search, so perhaps I overlooked something.

Thanks!

eudoxos, 2020-03-01:
I was able to do this by changing

https://github.com/juangburgos/QUaServer/blob/9f38ecf246d182bbf2c19cb44c607d8c225dc596/src/wrapper/quabasevariable.cpp#L253-L255

to

	
```c++
	UA_WriteValue wv;
	UA_WriteValue_init(&wv);
	wv.nodeId = m_nodeId;
	wv.attributeId = UA_ATTRIBUTEID_VALUE;
	wv.value.value = tmpVar;
	wv.value.hasValue = 1;
	if (sourceTimestamp.isValid())
	{
		QUaTypesConverter::uaVariantFromQVariantScalar(sourceTimestamp, &wv.value.sourceTimestamp);
		wv.value.hasSourceTimestamp = 1;
	}
	auto st = UA_Server_write(m_qUaServer->m_server, &wv);
```

with `const QDateTime& sourceTimestamp` being passed as an optional parameter to `setValue`.

The first attempt to implement `QUaBaseVariable::setSourceTimestamp` failed as `UA_Server_write` will reset the value if not provided (I hoped at first it will not touch it with `UA_WriteValue.value.hasValue=0`).

I am quite sure you won't be happy with adding an extra parameter to setValue ;) (there might be other meta-data someone might want to set with the value), you might have a better idea how to do it. Maybe breaking `setValue` into semi-internal prologue/epilogue methods handling type changes etc, and then setting the value itself in the middle.

juangburgos, 2020-03-02:
Hi, I can see the usefulness of setting custom timestamps. I have not required this so far but might have a use for it in the future.

Indeed, I would rather not add extra arguments to `setValue`, specially because, if we add an API to set custom timestamps, it might also be useful to be able to serialize such timestamps so they can be recovered after a server restart.

To serialize any attribute, `QUaServer` requires that attribute to be a `Q_PROPERTY`, then it can be serialized automatically by the current serialization API. So it would have to look something like this in [quabasevariable.h](https://github.com/juangburgos/QUaServer/blob/master/src/wrapper/quabasevariable.h):

```c++
class QUaBaseVariable : public QUaNode
{
	Q_OBJECT
	// Variable Attributes
	// other stuff...
	Q_PROPERTY(QDateTime sourceTimestamp READ sourceTimestamp WRITE setSourceTimestamp NOTIFY sourceTimestampChanged )
	Q_PROPERTY(QDateTime serverTimestamp READ serverTimestamp WRITE setServerTimestamp NOTIFY serverTimestampChanged )
	// other stuff...
public:
	// other stuff...
	QDateTime sourceTimestamp() const;
	void setSourceTimestamp(const QDateTime &sourceTimestamp);
	QDateTime serverTimestamp() const;
	void setServerTimestamp(const QDateTime &serverTimestamp);
	// other stuff...	
};
```

In the implementations of `setSourceTimestamp` and `setServerTimestamp` we would have to make sure that the value is not modified, only the respective timestamp.

Using this approach would also simplify visualization in GUI applications because, as in the serialization API, we can just iterate over the `QMetaPropery`'s.

Do you think you could try and give it a go? Currently I am busy with other stuff and might not be able to add new code to `QUaServer` in a couple of weeks.

eudoxos, 2020-03-10:
> In the implementations of setSourceTimestamp and setServerTimestamp we would have to make sure that the value is not modified, only the respective timestamp.

I don't know how to do that, to be honest... Apart from perhaps reading the value from the open62541 server, then setting with those respective metadata added. Not terribly efficient.

How about adding `QUaValueMetadata* metadata=nullptr` as arg to `setValue`? That would be future-proof in terms of passing more metadata if needed. Like:

```
struct QUaValueMetadata{
   QDateTime sourceTimestamp; // default-constructed is invalid and won't be used
   QDateTime serverTimestamp; // dtto
   // ...
}
```

By default, it would be `nullptr` and, even is used, only items explicitly set would be copied and activated in `UA_WriteValue`. Like this:

````
if (metadata){
   if(metadata->sourceTimestamp.isValid()) {
      QUaTypesConverter::uaVariantFromQVariantScalar(sourceTimestamp, &wv.value.sourceTimestamp);
      wv.value.hasSourceTimestamp = 1;
   }
   if(metadata->serverTimestamp.isValid()) {
      // ...
   }
   // handle other metadata ...
}
````

Does that sound sensible?

juangburgos, 2020-03-24:
I have made an implementation in the [alarms_conditions branch](https://github.com/juangburgos/QUaServer/tree/alarms_conditions) if you want to take a look. It won't be merged into master until I have implemented something meaningful in said branch though.

juangburgos, 2020-05-07:
Beware; breaking API changes in latest version. Now **browseName** is mandatory when cretaing any node as first argument. Then desired *nodeId* becomes second argument. See:

* <https://github.com/open62541/open62541/issues/3545>

commit eae03ab9 (2020-03-23): Working to fix #22
closed 2020-05-07 by juangburgos