QUaServer/QUaServer #23: Generating open62541 amalgation do not work any longer

derhexer13, 2020-06-27:
Hi,

the current verion do not generate the amalgation.
I'm using Windows 10 Pro 1909 / Visual Studio 2017 15.9.19

C:\QtProjects\QUaServer\src\amalgamation>"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvars32.bat"
**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.9.19
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'

C:\QtProjects\QUaServer\src\amalgamation>qmake -tp vc amalgamation.pro
Project MESSAGE: Compiling Open62541 for Windows
Project MESSAGE: CMake found.
Project MESSAGE: MsBuild found.
Project MESSAGE: Build directory created for open62541.
Project MESSAGE: Compiler Detected : MSVC++ 15.0 (Visual Studio 2017)
Project MESSAGE: Platform Detected : 32 bits
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.16.27035.0
-- The CXX compiler identification is MSVC 19.16.27035.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: C:/Programs/Python/Python37/python.exe (found version "3.7.3")
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.23.0.windows.1")
-- open62541 Version: 1.1.0-rc1-4-g1a667a05
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- Architectures included in amalgamation: win32;posix
-- The selected architecture is: win32
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE)
-- Could NOT find LATEX (missing: LATEX_COMPILER)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/QtProjects/QUaServer/depends/open62541.git/build
Project MESSAGE: CMake generate open62541 successful.
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
Build started 6/27/2020 12:07:35 PM.
Project "C:\QtProjects\QUaServer\depends\open62541.git\build\open62541.sln" on node 1 (default targets).
C:\QtProjects\QUaServer\depends\open62541.git\build\open62541.sln.metaproj : error MSB4126: The specified solution configuration "Debug|x86" is invalid. Please specify a valid solution configuration using the
Configuration and Platform properties (e.g. MSBuild.exe Solution.sln /p:Configuration=Debug /p:Platform="Any CPU") or leave those properties blank to use the default solution configuration. [C:\QtProjects\QUaS
erver\depends\open62541.git\build\open62541.sln]
Done Building Project "C:\QtProjects\QUaServer\depends\open62541.git\build\open62541.sln" (default targets) -- FAILED.

Build FAILED.

"C:\QtProjects\QUaServer\depends\open62541.git\build\open62541.sln" (default target) (1) ->
(ValidateSolutionConfiguration target) ->
  C:\QtProjects\QUaServer\depends\open62541.git\build\open62541.sln.metaproj : error MSB4126: The specified solution configuration "Debug|x86" is invalid. Please specify a valid solution configuration using th
e Configuration and Platform properties (e.g. MSBuild.exe Solution.sln /p:Configuration=Debug /p:Platform="Any CPU") or leave those properties blank to use the default solution configuration. [C:\QtProjects\QU
aServer\depends\open62541.git\build\open62541.sln]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:00.12
Project MESSAGE: Open62541 build successful.
The system cannot find the file specified.
Project ERROR: Failed to copy open62541 header file.

C:\QtProjects\QUaServer\src\amalgamation>

How do i solve this? 
Thx a lot for any hint!
Heribert

juangburgos, 2020-06-27:
Looking at your error code

error MSB4126: The specified solution configuration "Debug|x86" is invalid

It seems you are mixing 64 bits with 32 bits binaries somewhere.

derhexer13, 2020-06-27:
I do not know why i should mix the binaries.

I just opened a new cmd instance and started:
"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvars32.bat"
Result: [vcvarsall.bat] Environment initialized for: 'x86'

then i called:
qmake -tp vc amalgamation.pro
Result:
...
Project MESSAGE: Platform Detected : 32 bits
...
-- The selected architecture is: win32

juangburgos, 2020-06-28:
Are you using the special Qt command line or are you using a regular windows cmd? You should use the Qt one and make sure to have it match you compiler and call the correct vcvarsall.bat. That is all basic Qt handling. If you are not familiar with it maybe you should take a look at that first.

derhexer13, 2020-06-28:
It's a regular windows cmd. But the environment variables are set for successfully compiling qt sources.

The  vcvarsall.bat is replaced with vcvars32.bat (sets vs environment to x86).

Here is my build_qtopcua.bat that works fine inside a regular windows cmd.

```
echo Builds QtOpcUa
PAUSE
call "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvars32.bat"
cd C:\Qt\Qt5.14.2\5.14.2\Src\qtopcua
rmdir /s /q build
mkdir build
cd build
qmake ..
IF %ERRORLEVEL% NEQ 0 (
echo %ERRORLEVEL%
goto done
)
pause
nmake
IF %ERRORLEVEL% NEQ 0 (
echo %ERRORLEVEL%
goto done
)
nmake install
:done
```

Here is my build_quaserver.bat that shows the error.
echo Build QUaServer
```
call "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvars32.bat"
cd C:\QtProjects\QUaServer\src\amalgamation
qmake -tp vc amalgamation.pro
IF %ERRORLEVEL% NEQ 0 (
echo %ERRORLEVEL%
goto done
)
msbuild open62541.vcxproj /p:Configuration=Debug
msbuild open62541.vcxproj /p:Configuration=Release
:done
```

juangburgos, 2020-06-28:
I am sorry, but I cannot help you with custom build environments. I only support Qt's official cmd for building. Anything outside of that I don't know.

I have multiple projects both on Windows and Linux, 32 and 64 bits, using Qt cmd and they all compile fine. I recommend you to go that way.

Alternativelly you can build the open62541 amalgamation using Cmake as described by the open62541 docs. They can probably help you with that in their project. Then you link that amalgamation with QUaServer manually as well.

Good luck.

juangburgos, 2020-06-28:
I forgot. It seems form your error log that the VS solution does get generated, why not just open it, select a valid configuration and hit build.

commit dc761419 (2020-07-09): Fixes #23
closed 2020-07-09 by juangburgos