QUaServer/QUaServer #25: How to create arrays ?

christiangers, 2020-11-27:
Hello i am trying to learn about OPC UA.
I found your wrapper classes an want to use them to create a rapid prototype of an OPC server, and i have to define 
somthing like this:
:
QUaBaseDataVariable* alarmList;	
:
	alarmList = opcLevel3->addBaseDataVariable("alarmList");
	alarmList->setDataType(QMetaType::UShort);

but my alarm list contains 15 entries so the question is how to convert this into an array ? (UShort[15])
I can not find an example in the readme ...
Kind regards
Christian

juangburgos, 2020-11-27:
Hi, this currently works:

```c++
QUaFolderObject * objsFolder = server.objectsFolder();

auto arr = objsFolder->addBaseDataVariable("my_array");
arr->setValue(QVector<ushort>() << 1 << 2 << 3 << 4 << 5);
```

Note it is translated to OPC UA matrix type:

[image]

Hope is useful,

juangburgos, 2020-11-27:
BTW, you can also use a `QList<T>` instead of a  `QVector<T>`

christiangers, 2020-11-27:
Thanks a lot,
[image]
 so easy if you now how to do ...

closed 2020-11-27 by christiangers