QUaServer/QUaServer #26: User Type & Arrays of User Type

christiangers, 2020-11-27:
Hello,
my new problem is to create a custom type and to create an instance of it:

The header file:
----------------
#ifndef QUAAUTOSTARTREC_H
#define QUAAUTOSTARTREC_H

#include <QUaBaseObject>
#include <QUaBaseDataVariable>

class QUaAutoStartRec : public QUaBaseDataVariable
{
	Q_OBJECT

	Q_PROPERTY(QUaBaseDataVariable * active READ active)
	Q_PROPERTY(QUaBaseDataVariable * startTime READ startTime)

public:
	Q_INVOKABLE explicit QUaAutoStartRec(QUaServer* server);

	QUaBaseDataVariable* active();
	QUaBaseDataVariable* startTime();
};

#endif

The cpp file:
-------------
#include "quaVarTypeAutoStartRec.hpp" // moc header

QUaAutoStartRec::QUaAutoStartRec(QUaServer* server) :
	QUaBaseDataVariable(server)
{
	QUaBaseDataVariable* temp = this->active(); // <--- **This returns NULL !!!**
	this->active()->setDataType(QMetaType::Bool);
	this->active()->setValue(FALSE);
	this->startTime()->setDataType(QMetaType::UInt);
	this->startTime()->setValue(30000);
}

QUaBaseDataVariable* QUaAutoStartRec::active()
{
	return this->findChild<QUaBaseDataVariable*>("active");
}

QUaBaseDataVariable* QUaAutoStartRec::startTime()
{
	return this->findChild<QUaBaseDataVariable*>("startTime");
}

Parts of the caller file:
-------------------------
:
	server.registerType<QUaAutoStartRec>();
: 
	// test test test
	testRec = opcLevel3->addChild<QUaAutoStartRec>("testRec");
: 

The 'findChild' child method returns NULL !
I can not see what did i wrong ?
Please direct me to my fault ...

My first question is: 
Where is the instantiation of 'active ' and 'startTime' done ?
And my second question is:
How to create an array of a custom type ?
The vector method did not work here because the base object 
'QObject' is 'Q_DISABLE_COPY(QObject)' ?

thanx in advance

christiangers, 2020-11-27:
i forgot info question 2:
i tried:
QVector<QUaAutoStartRec> ASRVector(6, QUaAutoStartRec(server));

--> resulting compiler error : 
: 
D:\Quintec\Cpp2019\qt-everywhere-src-5.15.2\qtbase\src\corelib\kernel\qobject.h(467,5): message : 'QObject::QObject(const QObject &)': function was explicitly deleted
:

juangburgos, 2020-12-01:
Hi, 

Have you tried compiling and running **example 4 (types)** ? Why not try that out and see if it works for you, then modify it for your needs.

Also, you seem to be confused between custom *variable types* and custom *data types*. Please get yourself acquainted with OPC UA concepts of these are not clear for you. QUaServer supports creating custom *variable types*, whose value can be an array of a supported *data type*. But custom *data types* are not supported.

If you want indeed to create mutiple custom *variable type* instances, then you need to create them using the QUaServer API `addChild<T>`. You have to create them one by one and then you can add the returned *pointers* to an array for your own reference.

You can only have arrays of pointers for variable instances, because ownership of such instances belong to the QUaServer instance.

juangburgos, 2020-12-10:
Closed due to inactivity. Please re-open if problem persists.

closed 2020-12-10 by juangburgos