QUaServer/QUaServer #28: long int value cannot be set

eudoxos, 2021-04-07:
I would like to set a `long` value on a node but QVariant does not support `long` — it only works with `int` or `qlonglong` (and `qulonglong`).

Interestingly, passing `long long` actually does work if assertions are disabled; otherwise I get:
```
ASSERT: "this->dataTypeInternal() == m_dataType" in file ../QUaServer/src/wrapper/quabasevariable.cpp, line 388
```
I am not too much limited by this (the value is uptime in seconds) but it might be useful for someone to read it here, as the error is not very obvious.

eudoxos, 2021-04-07:
As a side note, [here is a detailed explanation](https://stackoverflow.com/a/589684/761090) on relationship between short, int, long, long long; the correspondence you assume is only valid for certain architectures (it is for amd64); the standard requires that e.g. `long` is at least 32bit, `long long` at least 64, so it is okay to map `UA_NS0ID_INT64` to `long long` (or `qlonglong`), though not necessarily vice versa (possibly narrowing conversion).

https://github.com/juangburgos/QUaServer/blob/a7f52ec1b6f7f862d315452f385f7c353c6e13eb/src/wrapper/quacustomdatatypes.cpp#L59-L66

juangburgos, 2021-04-23:
Thanks for the explanation, 

Maybe we can add the type exception in the `ASSERT` so it is not triggered in debug builds? Or do you propose changing the mapping in the `quacustomdatatypes.cpp`?

I wish I had time to dedicate to this now, sorry if I cannot address these issues promptly. My free time comes in every few months. Hope I get some time soon for fixes and features (would love to get x509 auth and optionset datatype support working).

juangburgos, 2021-08-06:
You could have set the `QMetaType::Long` or `QMetaType::ULong` explicitly in the `satValue` method which would map to `UA_NS0ID_INT64` and `UA_NS0ID_UINT64` respecitvely, but I added some template specializations for syntactic sugar.

commit 35fa343f (2021-08-06): Map both QMetaType::Long and QMetaType::LongLong to UA_NS0ID_INT64, same for unsigned (fixes #28)
closed 2021-08-06 by juangburgos