QUaServer/QUaServer #32: wrong object context in method invocation of QUaBaseObject subtype

ossi5150, 2021-07-06:
I tried example 4 (about types) where the TemperatureSensor is a subtype of QUaBaseObject.

I run into problems when I called the turnOn method from the client, the method is never called and the assertion message "Cannot call method on invalid C++ object." is shown.
Apparently, the object context casting failed (quaserver.cpp line 2346)
I look at the object context address apparently it is the QUaServer instance instead of the TemperatureSensor which should own the method.

juangburgos, 2021-07-06:
Cannot reproduce, see video below.

* Are you using latest QUaServer version with provided open62541 submodule version?
* Are you using any special compiler flags?
* Any useful information (platform, compiler, etc)?

[image]

ossi5150, 2021-07-06:
Sorry ofc,
platform is windows 
visual studio 2017
Qt 5.13
QUaServer (17th May 2021)

I do use a different open62541 than the provided one since I could not compile the QUaServer module with the error "typeIndex is not a member of UA_DataType"
I build it from the https://github.com/open62541/open62541
with the following configurations:
UA_BUILD_TOOLS
UA_ENABLE_AMALGAMATION
UA_ENABLE_DA
UA_ENABLE_DISCOVERY
UA_ENABLE_HISTORIZING
UA_ENABLE_METHODCALLS
UA_ENABLE_NODEMANAGEMENT
UA_ENABLE_PARSING
UA_ENABLE_SUBSCRIPTIONS
UA_GENERATED_NAMESPACE_ZERO_FULL

juangburgos, 2021-07-06:
Sorry, the README is clear on this:

[image]

QUaServer only supports the specific version form the provided submodule. Compiling cannot be any simpler `qmake -r -tp vc`, this also compiles the supported amalgamation.

ossi5150, 2021-07-06:
Thanks
so I did a fresh clone
and still has the same issue

https://user-images.githubusercontent.com/87018676/124623370-3c112680-de7c-11eb-9a25-50ec3b8a983d.mp4

juangburgos, 2021-07-06:
Are you using now the amalgamation provided by QUaServer, or still your own one from the original open62541?

ossi5150, 2021-07-06:
I'm using the one QUaServer provided now

juangburgos, 2021-07-06:
That is weird, because I am using almost the same as you (VS 2017, Qt 5.12), I have even tested on Linux and all works. Can you think of anything else you might have different?

juangburgos, 2021-07-06:
Is this [https://github.com/FreeOpcUa/opcua-client-gui](https://github.com/FreeOpcUa/opcua-client-gui) the client you are using to test?

ossi5150, 2021-07-06:
yes that the one I use

juangburgos, 2021-07-06:
OMG I wanted to try it but the WinPython dependencies are huge, have you tried with UaExpert?

ossi5150, 2021-07-06:
Thanks, I tried with UaExpert and it works just like your video

juangburgos, 2021-07-06:
Damn, so the `opcua-client-gui` client appears to do something incorrectly. When I manage to run it, I will check if something can be done.

ossi5150, 2021-07-06:
Yea, it seems to be the case.
Thanks for the quick answer!

closed 2021-07-06 by ossi5150