QUaServer/QUaServer #34: linking with mingw32-w64: undefined to references

eudoxos, 2021-09-22:
Hi, I see this (a bit unexpected and arcane) issue when linking with QUaServer (latest) on mingw32-w64 platform:
```
/usr/bin/x86_64-w64-mingw32-ld: QUaServer/libQUaServer_static.a(quaoptionsetvariable.cpp.obj):quaoptionsetvariable.cpp:(.rdata$.refptr._ZN16container_traitsI5QListI16QUaLocalizedTextEE7arrTypeE[.refptr._ZN16container_traitsI5QListI16QUaLocalizedTextEE7arrTypeE]+0x0): undefined reference to `container_traits<QList<QUaLocalizedText> >::arrType'
/usr/bin/x86_64-w64-mingw32-ld: QUaServer/libQUaServer_static.a(quaoptionsetvariable.cpp.obj):quaoptionsetvariable.cpp:(.rdata$.refptr._ZN16container_traitsI5QListIbEE7arrTypeE[.refptr._ZN16container_traitsI5QListIbEE7arrTypeE]+0x0): undefined reference to `container_traits<QList<bool> >::arrType'
```
I cannot produce a MWE (the code is quite big), but maybe it rings the bell for you immediately as to what the cause could be? If not, leave it, I will figure it out at some point, hopefully.

juangburgos, 2021-09-22:
I only use GCC on Linux, and indeed it is much picky with templates. 

What Qt version and `g++ -v` do you have?

It probably has a problem with [this `valueInternal`](https://github.com/juangburgos/QUaServer/blob/ebc1ae525bee3fbfa3eaf3a83d053f7231bbcda2/src/wrapper/quabasevariable.h#L262) overload.

If you are not using the OptionSet type (I think last time [I couldn't even make it work](https://github.com/open62541/open62541/issues/3934)), you can comment it out in [`quaserver.pri`](https://github.com/juangburgos/QUaServer/blob/ebc1ae525bee3fbfa3eaf3a83d053f7231bbcda2/src/wrapper/quaserver.pri#L67).

If you find a more elegant solution, let me know or hit me with a PR.

juangburgos, 2021-09-22:
Maybe try to add `const QList` and `const QVector` specializations in [`quabasevariable.h`](https://github.com/juangburgos/QUaServer/blob/ebc1ae525bee3fbfa3eaf3a83d053f7231bbcda2/src/wrapper/quabasevariable.h#L12)?

juangburgos, 2021-09-22:
Oh wait, yes, I did have the same in Linux, so maybe just add a preprocessor condition for GCC on Windows [**here**](https://github.com/juangburgos/QUaServer/blob/ebc1ae525bee3fbfa3eaf3a83d053f7231bbcda2/src/wrapper/quabasevariable.h#L255)

eudoxos, 2021-09-22:
I use gcc 9.3 on Linux, the mingw32-w64 cross-compiler is more recent:
```
x86_64-w64-mingw32-gcc (GCC) 10-win32 20210408
```
This is not really a compiler issue (the code compiles, and gcc is quite conforming to the standard) but a linker issue. As I read the message (and I have probably less understanding of template-fu than you), it is `arrType` which is not known; that is a `static const` inside a templated class. I am not sure how are those allocated — normally static members need explicit instantiation so that they have a defined storage, right?. Perhaps `constexpr` could be used instead of `static const` for `arrType`?

juangburgos, 2021-09-22:
You can check if `constexpr` works.

The `container_traits` struct is just there to try and deduce if the user passes a `QList` or `QVector` and the inner type. VS2013 and forward has no sweat on these types of expressions. 

The ideal solution would be [to use template-template parameters](https://stackoverflow.com/questions/213761/what-are-some-uses-of-template-template-parameters), but that seems supported only starting C++17 (and I need C++11 compat for my projects).

PS: I also don't like the template syntax at all. But I love more a simple user-facing API than I hate the template syntax.

eudoxos, 2021-09-22:
> You can check if `constexpr` works.

I just tried and it works :) `static constexpr`. Does it break your compatibility requirements? If not, I will do a PR.

commit b8041706 (2021-09-23): Hopefully fixes #34
closed 2021-09-23 by juangburgos