QUaServer/QUaServer #41: Qt6 port available

ArsMasiuk, 2023-01-24:
Hello,
here is my port of QUaServer library to Qt6: https://github.com/ArsMasiuk/QUaServer-Qt6
It has been compiled with Qt 6.4.2 and Visual Studio 2022.

jojosula, 2023-02-24:
Hi,
I found a issue on Qt6.  Qt6 variant canConvert<QVariantList> result is different from Qt5 when variant type is QString and QByteArray.
Here is the canConvert sample code. You can run it with Qt5 and Qt6 to see the result.
https://gist.github.com/jojosula/0889e5190eae04cbc04819913f539f34

Here is my fork to fix this issue.
https://github.com/jojosula/QUaServer

juangburgos, 2023-02-24:
Hi, thank you for sharing. I am not currently activelly contributing to this repo, but if you rebase your fork with respect to this one and create an PR, I'd be happy to make you a member of the organization (with certain admin rights). Cheers.

closed 2023-05-17 by juangburgos