QUaServer/QUaServer #44: Multiple output in the addMethod function

GeoCode0, 2023-05-09:
Hi, how can i create a method with more than one output. Do i have to build something myself to do so or there is something i'm missing?

juangburgos, 2023-05-10:
Currently not implemented, although possible if defining an output type that supports multiple values, like a `std::tuple`. 

My guess is that it might require some template meta-programming magic.

Sounds like an interesting feature. In case somebody is interested in implementing it, I am willing to review and merge it.

GeoCode0, 2023-07-10:
Is it possibile to use references on the user input when he uses a method as an "extra return" to cheat the issue?

juangburgos, 2023-07-10:
> use references on the user input

Not possible, but an alternative to what you want is the one described in the main `README.md`, namely:

> Using the Lambda Capture it is possible to change Objects or Variables:

```c++
auto varNumber = objsFolder->addBaseDataVariable();
varNumber->setDisplayName("Number");
varNumber->setValue(0.0);
varNumber->setDataType(QMetaType::Double);

objsFolder->addMethod("incrementNumberBy", [&varNumber](double increment) {
	double currentValue = varNumber->value().toDouble();
	double newValue = currentValue + increment;
	varNumber->setValue(newValue);
	return true;
});
```

I.e. just create a separate object that represents the output and let the method update that object.