QUaServer/QUaServer #46: DEFINES += QT_NO_CAST_FROM_ASCII

sergey-kuzminov, 2023-06-20:
Hi. If we add the string DEFINES += QT_NO_CAST_FROM_ASCII to the .pro file, we will get a lot of warnings. To get rid of these warnings, you need to replace QString with QStringLiteral/QString::fromLatin1/QString::fromUtf8

Here's an example, but I'm not sure about everything: 
[QUaServer-QT_NO_CAST_FROM_ASCII.zip](https://github.com/QUaServer/QUaServer/files/11800231/QUaServer-QT_NO_CAST_FROM_ASCII.zip)

juangburgos, 2023-06-20:
Thanks for the info
Contributions are only accepted through PRs. But looking at the sources you put here, your changes would have to be inside a proper `#ifdef QT_NO_CAST_FROM_ASCII `.

sergey-kuzminov, 2023-06-21:
This patch changes QString constants to QStringLiteral.
[QStringLiteral.zip](https://github.com/QUaServer/QUaServer/files/11816286/QStringLiteral.zip)

closed 2023-06-23 by juangburgos

sergey-kuzminov, 2023-06-23:
Unfortunately PR closes only part of the problem. There will be another PR in the future.

PR #47 (2023-06-23): Replacing QString constants with QStringLiteral.
Fixes #46