QUaServer/QUaServer #58: Invalid type names.

sergey-kuzminov, 2023-07-26:
There are incorrect type names in the QUaServer library. For example QStringLiteral("Bool"), but it should be QStringLiteral("bool"). I don’t know what it affects and how to check the performance of this part of the code, so I don’t fix it. Here is a list of incorrect type names and their correct values:

`qDebug() << QMetaType::type("Bool")     << QMetaType::type("bool")        << QMetaType(QMetaType::Bool).name();
    qDebug() << QMetaType::type("Char")     << QMetaType::type("char")        << QMetaType(QMetaType::Char).name();
    qDebug() << QMetaType::type("SChar")    << QMetaType::type("signed char") << QMetaType(QMetaType::SChar).name();
    qDebug() << QMetaType::type("UChar")    << QMetaType::type("uchar")       << QMetaType(QMetaType::UChar).name();
    qDebug() << QMetaType::type("Short")    << QMetaType::type("short")       << QMetaType(QMetaType::Short).name();
    qDebug() << QMetaType::type("UShort")   << QMetaType::type("ushort")      << QMetaType(QMetaType::UShort).name();
    qDebug() << QMetaType::type("Int")      << QMetaType::type("int")         << QMetaType(QMetaType::Int).name();
    qDebug() << QMetaType::type("UInt")     << QMetaType::type("uint")        << QMetaType(QMetaType::UInt).name();
    qDebug() << QMetaType::type("Long")     << QMetaType::type("long")        << QMetaType(QMetaType::Long).name();
    qDebug() << QMetaType::type("LongLong") << QMetaType::type("qlonglong")   << QMetaType(QMetaType::LongLong).name();
    qDebug() << QMetaType::type("ULongLong")<< QMetaType::type("qulonglong")  << QMetaType(QMetaType::ULongLong).name();
    qDebug() << QMetaType::type("Float")    << QMetaType::type("float")       << QMetaType(QMetaType::Float).name();
    qDebug() << QMetaType::type("Double")   << QMetaType::type("double")      << QMetaType(QMetaType::Double).name();

    qDebug() << QMetaType::type("qlonglong")  << QMetaType::type("long long");
    qDebug() << QMetaType::type("qulonglong") << QMetaType::type("unsigned long long");`

juangburgos, 2023-07-26:
Can you point to specific parts pf the code where these are defined/used? I cannot recall from the top of my head.

sergey-kuzminov, 2023-07-26:
https://github.com/QUaServer/QUaServer/blob/0d0e5c2c6bfedf3ca921bb1c299050114cee083c/src/wrapper/quacustomdatatypes.cpp#L20
https://github.com/QUaServer/QUaServer/blob/0d0e5c2c6bfedf3ca921bb1c299050114cee083c/src/wrapper/quacustomdatatypes.cpp#L125

sergey-kuzminov, 2023-07-26:
It seems these type names are for serialization. I didn't understand it right away.

closed 2023-07-26 by sergey-kuzminov